Wren6991/PicoDVI #1: usb support?

cleverca22, 2021-01-21:
https://www.phoronix.com/scan.php?page=news_item&px=Linux-Generic-USB-Display

this is a fully open spec on how to drive a display over usb, given that the 2nd core is entirely unused, i'm guessing it has enough processing power to be a usb gadget, and implement that protocol, which would then expose the framebuffer to a host over usb!

linux already has drivers to act as either end of that protocol, and there are examples of running it on a pi0 in gadget mode

Wren6991, 2021-01-21:
Note the second core is idle when displaying a static pixel-doubled image, but many of the examples use both cores. [See here](https://github.com/Wren6991/picodvi/tree/master/software). I'll put up some YouTube videos at some point.

This would be a neat example, though pretty limited in resolution. I'd rather work on getting the DVI code imported into the `scanvideo` stack (currently experimental), so that you can use the same example with e.g. VGA, DBI. It wouldn't be developed in this repo. Note we are all quite busy around launch, so this isn't going to happen for a while.

Wren6991, 2021-01-21:
Sorry I was a bit terse and forgot to preface this with: yes this is absolutely a thing that should exist :) and we have talked internally about that very driver in the context of the VGA code

raszpl, 2021-01-23:
USB 1.1 =12Mbit = ~1MB/s

cleverca22, 2021-01-24:
@raszpl the protocol supports things like RGB565 (16bpp) and LZ4 compression of the stream, along with partial updates of the framebuffer

usb1 will definitely harm the fullscreen refresh rate, but it still looks feasible, but obviously not a high priority

fanoush, 2021-01-25:
Slightly related, could PIO do additional USB in software? Even AVR ATtiny could bitbang it.
Too bad there is nothing in USB between fullspeed (12Mbps) and high speed 480Mbps (as the high speed mode is probably too high).

cleverca22, 2021-01-28:
@fanoush if you use a high-speed usb phy, with an 8bit bus driven at 60mhz (well within pio reach), then you can definitely get highspeed usb on an rp2040

but you would have to re-implement the entire usb controller in software

fanoush, 2021-01-29:
@cleverca22 yes, but then it may be easier to use the idea I mentioned in the forum - use cheap small microsd reader and emulate sd card to that reader. mmcplus/emmc even has 8 data pins but even plain SD 4 bits@100MHz would be pretty fast. Then you could do similar protocol like uf2 to communicate with it over usb storage.

But even another full speed (12MBits) USB in software could be interesting so it could be usb host and client at the same time.