Wren6991/PicoDVI #4: Question: Could this be adapted to display at higher resolutions at the cost of fps/fill rate?

MatinatorX, 2021-02-07:
Awesome project you have here! Got me thinking about other use cases which don't require smooth motion, such as a cheap photo-frame controller or even a cheap DLP 3D printer controller. Even if it took a few seconds to stream in the image off of NAND and fill the display, for the cost it would still be very compelling!

From what I have read, some HDMI receiver ICs really don't like anything below 24Hz, but many are happy down to 15Hz and possibly below (I haven't found any tests below 15Hz). Would be very interesting to see max resolution that could be attained within the bandwidth the RP2040 allows if FPS doesn't matter. 4K even at 0.5 fps would be nuts.

I just got my order of Pico dev boards and will be cobbling together a test bed for your project ASAP, again great work!

Side note: Have also been thinking about MIPI DSI support (8 PIOs vs 8 lane MIPI DSI?) but that seems like a pipe dream, especially since publicly available information on the spec seems almost nonexistent...

Wren6991, 2021-02-08:
The system clock needs to equal the TMDS bit clock, in order to physically shovel the bits out of the chip fast enough. Even at the lowest permissible DVI pixel clock of 25 MHz, this results in a 250 MHz system clock, which is already quite spicy. 640x480p60Hz requires a 25.2 MHz pixel clock. This seriously limits the maximum resolution of scanout!

Wren6991, 2021-02-08:
Also note that the full-resolution VGA example (`software/apps/vista`) is streaming the entire image from flash every frame, because 640x480 RGB565 is more than double the size of the on-chip RAM (it's around 600 kB)

MatinatorX, 2021-02-08:
Thank you for the response! However, I think I might not have explained myself very well.

I was curious if a 4K image could be displayed at the same DVI pixel clock of 25MHz, just at a much, _much_ lower framerate. Doing the math, assuming the same bpp, displaying 4K images at 25.2MHz pixel clock rate would result in a max framerate of around 2.2 fps, versus the 60 fps of the demo at 640x480. This would be more than sufficient for something like a digital picture frame switching images every 30 seconds, or a DLP printer spending a few seconds on each layer.

I could be completely missing a critical part of how this works, though! I had thought there was no relationship between the pixel clock and the resolution and that it was more of a three-way relationship balancing resolution and FPS at that clock. I will continue to research to try and fill the gaps in my knowledge.

Wren6991, 2021-02-08:
Oh, I didn't realise that by framerate you meant the actual scanout rate out to the monitor, as opposed to repeating frames! Um, you can try, I suspect you will get a nice "Out Of Range" on your monitor :)

There are lists of permissible timings (you can find one such list by googling for "CEA 861D PDF") and there's nothing close to 2 FPS at any resolution.

closed 2021-05-17 by Wren6991