Wren6991/PicoDVI #5: HDMI Audio?

tmzt, 2021-02-11:
Is there any way to support HDMI Audio output with this, maybe with mostly static graphics (now playing or album art)?

Wren6991, 2021-02-11:
It should be possible but I haven't implemented it.

kargeor, 2021-02-12:
I would love to have audio. Maybe I can help implement it?

blueOkiris, 2022-03-01:
@Wren6991 I'd like to look into the audio aspect, at the very least for my own projects.
Do you have a link to a reference where you got the info for the DVI/HDMI communications protocol?

mlorenzati, 2022-06-13:
I second the proposal to the HDMI audio, also where could we find information about technical implementation.

mlorenzati, 2022-06-13:
> @Wren6991 I'd like to look into the audio aspect, at the very least for my own projects. Do you have a link to a reference where you got the info for the DVI/HDMI communications protocol?
@blueOkiris check https://www.intel.com/content/www/us/en/docs/programmable/683798/22-1-19-7-0/hdmi-overview.html
basically on the actual video stream, it has to be placed an additional audio frame (data island preamble-guard-audio-guard-video preamble)

c1570, 2023-02-13:
https://github.com/shuichitakano/pico_lib (apparently a C++ fork of PicoDVI) has audio support.

mlorenzati, 2023-03-03:
> https://github.com/shuichitakano/pico_lib (apparently a C++ fork of PicoDVI) has audio support.

I'm taking this info to add audio support to a project based on Wren's pico lib, it will be opensource.
For the time being I added something from there that was missing in current implementation which is to be able to start and stop DVI transmission without breaking.

I use it since I built an RGB2HDMI converter (for retro computers) and when there is no RGBHV signal I can stop HDMI after a while and allow HDMI screen to turn off.

mlorenzati, 2023-03-11:
@c1570 @Wren6991 I took the work on pico_lib From @shuichitakano and tried to honor the same flows.
DVI start stop works and DVI works if dataislands are not enabled.
Anyone wanting to collaborate?
https://github.com/mlorenzati/PicoDVI

mlorenzati, 2023-03-20:
The work can be merged back here from https://github.com/mlorenzati/PicoDVI, kudos to @shuichitakano