Wren6991/PicoDVI #6: Assembly troubleshooting?

paulcombe, 2021-03-02:
I ordered a few boards from OSHPark and put them together, but I'm not getting sync on hello_DVI, even with a 1.3V overvolt. I don't have an oscilloscope; is there a good way to troubleshoot assembly? Some macro photos look like every resistor is set properly (none fall off, at least), and there are no bridges on the plug pins, but I'm not confident. With a scope, I'd just look at the outputs, but don't got. Anyone got a clever idea I can borrow?

Alternatively, would there be any gotchas just making the board a half inch longer and using 0603 resistors for a bit more breathing room?

Wren6991, 2021-03-08:
I guess you could buy one of these awful HDMI terminal block things if you want to confirm connectivity between RP2040 and the socket

https://www.amazon.co.uk/CERRXIAN-Terminal-Solderless-Breakout-Connector/dp/B083DL7RVY

increasing the footprint size to 0603 imperial would probably be fine

Wren6991, 2021-03-08:
Here is a known-good binary for a PicoDVI Rev C board, so you can rule that out too.

[hello_dvi_picodvi_revc.zip](https://github.com/Wren6991/PicoDVI/files/6104455/hello_dvi_picodvi_revc.zip)

This one also blinks the LED (GPIO21) so you can confirm the program is running and not crashing. You should not need 1.3V.

paulcombe, 2021-03-13:
Ah, you know what would have really helped? Specifying that I was using the DVI Sock, because the default pinout for the TMDS lines is different.

I changed the main.c to use the sock pinouts, and I have an improvement: My monitor says something is plugged in but not giving a signal. This is an improvement over "nothing is plugged in"! I'll look at my soldering job again.

paulcombe, 2021-03-23:
Oh, follow-up for closure: ultimately the resistors and board were fine. The issue was pins bridging under the HDMI connector because solder wicked up during drag soldering. Ultimately, I wasn't able to fix those. I'm guessing there's more technique for these than there is for plastic packaged ICs, and I will look that up.

Left open in case anyone has a hot soldering tip and trick, but issue closed. PEBKAC.

Wren6991, 2021-05-17:
Dousing with flux and/or alcohol then swiping with the soldering iron is helpful for clearing shorts on fine-pitch connectors, in case you want to give it another go! The boiling alcohol helps to dislodge the bridge.

actclekcz, 2024-12-23:
> Here is a known-good binary for a PicoDVI Rev C board, so you can rule that out too.
>
> [hello_dvi_picodvi_revc.zip](https://github.com/Wren6991/PicoDVI/files/6104462/hello_dvi_picodvi_revc.zip)
>
> This [...]

Thank you, this helps a lot!