Wren6991/PicoDVI #9: Will this work with MicroMod Big Display Carrier Board

ralphjy, 2021-03-11:
Hi Luke,
This isn't specifically your issue, but I was wondering if you have any experience or knowledge of the Sparkfun MicroMod Big Display Carrier Board and RP2040 MicroMod.  I recently bought this setup and the example uf2 files they provide don't appear to work.  No responses from their forum as yet.
I only ask because they reference your software and their design is somewhat based on your design (I noticed they still use the capacitive coupling of the HDMI signals).
I've tried compiling your software and it doesn't work on this setup either (compiles fine, but no display).
Any suggestions for troubleshooting?  RP2040 seems to load okay, I'll try to get a look at the signals to the HDMI connector with a scope.
[MicroMod_Big_Display_Schematic.pdf](https://github.com/Wren6991/PicoDVI/files/6124210/MicroMod_Big_Display_Schematic.pdf)

Thanks,
Ralph

Link to their board https://www.sparkfun.com/products/17718

Wren6991, 2021-03-11:
I guess you have already added this to the top-level CMakeLists.txt?

```
add_definitions(-DDEFAULT_DVI_SERIAL_CONFIG=micromod_cfg)
```

Here is a UF2 I just built and tested using that pin config, works for me:

[hello_dvi_micromod.zip](https://github.com/Wren6991/PicoDVI/files/6124660/hello_dvi_micromod.zip)

You may want to try a different monitor. Protocol-wise the DVI output should be fine, but some monitors might be unhappy with the logic levels from the capacitively coupled output circuit, or with the fact that 5V is not applied to the HDMI 5V input (strangely the board seems to treat that pin as a 5V output).

ralphjy, 2021-03-11:
Hi Luke,
Thanks for the quick response.  I just tried the UF2 that you built and it also works for me!

The Sparkfun repo only had a couple of example UF2 files, but no source code.  Those files did not work.

Also, when I compiled your code, I did not change the add_definitions and it also looks like that line was commented out.
# add_definitions(-DDEFAULT_DVI_SERIAL_CONFIG=picodvi_pmod0_cfg)

I will give that a try later today and recompile your code.

Great to see this work.  Really appreciate that you did this project.

ralphjy, 2021-03-11:
Actually, I have to take back some of what I said.  The two example files from the Sparkfun repo do work.  I think I got tricked because of what you mentioned about monitors possibly being the issue.  I had tried 2 different monitors and also both HDMI and DVI inputs and 2 different cables with nothing working.  When I tried your UF2 file, it worked initially but after I tried recompiling your code - nothing worked again, even the working UF2 file.  I realized that when I first tried your file that I had swapped an HDMI output from an RPi4 with the HDMI output from the RP2040.  I guess the monitor did not try to re-validate the HDMI input when I did the swap.  I verified that I could get other UF2 files to work if I started with the RPi4 signal and then quickly swapped to the RP2040.  

Also, it looks like the capacitively coupled outputs may not be working that well in my setup.  The Vista example gives me a mostly white screen.  The Terminal example seems to not sync correctly and the Moon example has a lot poorer resolution that you had.
[image]
[image]

Do you think that it would be worth swapping the caps for resistors?  The small parts will be challenging.

Wren6991, 2021-03-11:
>  The Vista example gives me a mostly white screen

You need to load the images as well as the viewer -- drag the `vista_data.uf2` file from the `assets/` directory. It'll take a a while to program.

> the Moon example has a lot poorer resolution that you had

If you read the source you will see there are multiple resolutions you can select

ralphjy, 2021-03-12:
Thanks.  All makes sense when you know what you are doing.  I'll start looking at the code before asking questions next time.  I guess I was looking for instant gratification :^)

All working now except for the start up issue with the monitors.  I'll take a look at that 5V input that you mentioned.  I've tried a couple of monitors (and cables) and even tried the DVI input on one of them.  I'll try another one just for fun.