Wren6991/PicoDVI #12: DVI and MicroPython support

mchobby, 2021-05-16:
Is there any way being able to have DVI output from MicroPython script (eg: create a terminal, having REPL available on screen, display some graphics) ?

Wren6991, 2021-05-17:
You can take a look at this fork where I hacked in a virtual console with DVI:

https://github.com/Wren6991/micropython/tree/dvi-vt100

It's not for the faint of heart, and IIRC I broke a few things (like flash programming). The code is there if you want to hack on it.

The approach was to steal the second core from MicroPython, and run some bare metal DVI stuff on core 1, with the MicroPython interpreter on core 0.

AtomiJD, 2021-08-20:
I followed your approach to integrate DVI in Micropython. 

Yes its not so easy/possible to get flash programming running while DVI is running. I stopped trying this because of some notes from flash.h:

"* Like other hardware_flash functions, the flash is not
 * accessible for execute-in-place transfers whilst the command is in
 * progress, so entering a flash-resident interrupt handler or executing flash
 * code on the second core concurrently will be fatal. To avoid these pitfalls
 * it is recommended that this function only be used to extract flash metadata
 * during startup, before the main application begins to run"
 
After some hours of trying the flash thing, I decided to store the data on SD.
Changing the frozen modules _boot.py for SD Card Support and adding the sdcard.py
source and do some minor changes on main.c everything works fine (vscode/rshell/Thonny)

Setup:
DVI Breakout Board	on pins 12, 18, 16, clk pin 14,
MCP4822 DAC		on SPI 1
SD Card Reader		on SPI 0
Pico Probe 		on UART GPIO 0,1

DVI uses 
- Front/Back buffer with 320x240 8bpp, nice for gaming
or
- 1 Buffer 320x240 16bpp
with blinking cursor, 8x8 font, etc.

The DAC is for sound generation has 8 channels sine/saw/triangle/rectangle/noise generation
with ADSR envelope and 12 Bit sound at 22050 sample rate.

The Micropython core modules are enhanced for vga, sprite, graphic and sound.

Finally I have a nice retro computer running Python. 
Thanks a lot for your work on that DVI thing.

jblanked, 2025-04-06:
> I followed your approach to integrate DVI in Micropython.
>
> Yes its not so easy/possible to get flash programming running while DVI is running. I stopped trying this because of some notes from [...]

Have you released this somewhere?