Wren6991/PicoDVI #16: Reverse 1bpp

fruit-bat, 2021-08-04:
Please could we have a version of 1bpp that emits the pixels from each byte in the reverse order.
Many thanks for publishing this library!

melkael, 2021-08-19:
Have you tried changing the invert_diffpairs parameter for your board in this file ?

https://github.com/Wren6991/PicoDVI/blob/master/software/include/common_dvi_pin_configs.h

When i dit so it showed the negative of the original image on screen.

Wren6991, 2021-08-20:
> Please could we have a version of 1bpp that emits the pixels from each byte in the reverse order.
> Many thanks for publishing this library!

Implementation is here:

https://github.com/Wren6991/PicoDVI/blob/7423e4d62e92feeb0f4ad2027f1c18c2600a737b/software/libdvi/tmds_encode.S#L193-L251

You would need to:

1. Reverse the order of the lookup table `tmds_1bpp_table`, to reverse the bits within each half-byte
2. Swap the two shift instructions for each invocation of `tmds_encode_1bpp_body`, for example `lsls 3  lsrs 1` becomes `lsrs 1 lsls 3`, to reverse the half-bytes within each byte

fruit-bat, 2021-08-21:
Many thanks for getting back to me on this. I will try it out as as soon as
I get a chance.

I've started messing about with an emulator (not original, I know) for an
old computer... https://github.com/fruit-bat/pico-sorcerer-2. The display
is rock solid, amazing for a microcontroller.

Cheers,

Phil

Wren6991, 2021-08-22:
That's a cool project! I added a new config `DVI_1BPP_BIT_REVERSE` with 844c6197d.

You should just be able to add `target_compile_definitions(app_name_here PRIVATE DVI_1BPP_BIT_REVERSE=1)` to your CMakeLists, please let me know if it works.