Wren6991/PicoDVI #19: Red bars on screen

Blackswirllife, 2021-09-09:
Hey i have a issue and i dont get it why this happens.

I use a DVI Sock for my test and the pico_sock_cfg config but when im compile and uplode to my Pico i get Red Bars
i use the apps whit no Editing to try but when im using the demo programms from BuyZero this works fine.

Anyone have a idea why this happens ?

Wren6991, 2021-09-10:
Did you build with `cmake -DPICO_COPY_TO_RAM=1` like in the readme?

Blackswirllife, 2021-09-10:
> Did you build with `cmake -DPICO_COPY_TO_RAM=1` like in the readme?

I use VS Code but i have the problem solved now.

i has use debug mode to compile but with release mode it's working.

thanks for the answer

closed 2021-09-10 by Blackswirllife

Wren6991, 2021-09-12:
Glad it's fixed.

The red bars are a placeholder for when the core was late with rendering or encoding a scanline. They indicate there's some issue with the build that's making things slower than they should be.