Wren6991/PicoDVI #20: pixelHEIGHT 1:1

sabsun77, 2021-09-14:
Hi,
//I use your code in Arduino IDE

Code to improve pixelHEIGHT ratio 1:1 please;
[https://youtu.be/eJBpgg-vlX8](url)
Thanks,

Wren6991, 2021-09-20:
What is the change you want from this repo?

DVI_SYMBOLS_PER_WORD is set based on the encoder you are using, almost all of them use 2 symbols per word, which is more efficient than 1. It's not something you can change freely. If you set it to the wrong value you will get a really screwed up signal, which monitors are not obligated to display at all.

Wren6991, 2021-09-20:
Specifically it's the full-resolution RGB encoder that requires `DVI_SYMBOLS_PER_WORD=1`, for all others the correct value is the default of 2.

sabsun77, 2021-09-20:
The sample code doesn't have any problems. but want  apply to my screen

In Display HDMI 800*480
I'm set  DVI_SYMBOLS_PER_WORD 1

#define FRAME_WIDTH 800
#define FRAME_HEIGHT 480
#define VREG_VSEL VREG_VOLTAGE_1_30
#define DVI_TIMING dvi_timing_800x480p_60hz
 
I custom sprites 800* 80 draw button w=100 h=80  and encoder 6 loop =FRAME_HEIGHT 480

 in code "tmds_encode_data_channel_16bpp"

tmds_encode_data_channel_16bpp(scanbuf, tmdsbuf + 0 * words_per_channel, pixwidth/DVI_SYMBOLS_PER_WORD  , DVI_16BPP_BLUE_MSB,  DVI_16BPP_BLUE_LSB );
  tmds_encode_data_channel_16bpp(scanbuf, tmdsbuf + 1 * words_per_channel, pixwidth/DVI_SYMBOLS_PER_WORD  , DVI_16BPP_GREEN_MSB, DVI_16BPP_GREEN_LSB);
  tmds_encode_data_channel_16bpp(scanbuf, tmdsbuf + 2 * words_per_channel, pixwidth /DVI_SYMBOLS_PER_WORD , DVI_16BPP_RED_MSB,   DVI_16BPP_RED_LSB  );

but , pixel width is show 800 and  pixel height is show 240px  is not show 480px
Is there a way to make the height of the pixels show the full FRAME_HEIGHT ?
https://1drv.ms/u/s!AgM7p_3BoHejlnybbUUja0lsPTrG?e=UpPVaO

sabsun77, 2021-09-24:
> > I'm set DVI_SYMBOLS_PER_WORD 1
> 
> Why did you set that? What did you want it to do?
I want to show FRAME_HEIGHT 480 equal to dvi_timing.v_active_lines = 480 if possible.

closed 2022-06-13 by Wren6991