Wren6991/PicoDVI #27: Release one core in 640x240 mode

mlorenzati, 2022-04-23:
I successfully integrated pico dvi to capture RGB15Khz video (pre VGA) with a triple s&h ADC.
The 320x240 mode leaves 1 core for my HSYNC VSYNC interrupts and to call DMA which calls dedicated pio.
The 640x480 is not working (either capture and pause hdmi or viceversa) since 2 cores are fully used and interrupts are interferring in bad moments.
Since RGB15K uses just 240 vertical lines in progressive mode, could be the code arranged to either generate a black line in odd lines or to repeat the odd line in the even ones.
How could this be achieved?
[image]

Wren6991, 2022-04-24:
You could start by increasing DVI_VERTICAL_REPEAT to 2, so that the horizontal interrupt uses each TMDS buffer for two scanlines. Search for it in the code, I think 2 is the default, and the 640x480 demos override it to 1 in their cmake files.

mlorenzati, 2022-04-24:
Excellent, I tested it and now I have a free core to attend IRQs and do some bg tasks!
Thanks @Wren6991 for your reply, of course when and if this sees the light there would be proper mention to you and will commit to the BSD-3 clause license.

closed 2022-04-24 by mlorenzati

mlorenzati, 2022-04-25:
[image]
640x240 RGB 2 HDMI, I run a half buffer with circular indexes.

cnlohr, 2025-06-04:
Could you provide more information on this? What speed is the DMA running at in 640x240 mode?