Wren6991/PicoDVI #31: 640x480@50hz

mlorenzati, 2022-06-13:
Is it possible an valid to output 640x480 at 50hz?
I built an RGB2HDMI scanner and the limitation is the  memory to store all the video.
If it could be possible to synchronize RGB timings to hdmi timings, I would only need just a few width lines to capture and then render to hdmi.

Wren6991, 2022-06-13:
If you scaled the regular 640x480 60Hz mode down to 50 Hz then the pixel clock would be too slow for the DVI spec (minimum 25 MHz) and most monitors won't accept it. You might be able to design a 640x480 CEA mode that some monitors woluld accept.

Wren6991, 2024-03-21:
Experimentally this seems to work: just add additional hblank to the existing 480p60 mode to increase frame time by 1/5th.

```
diff --git a/software/libdvi/dvi_timing.c b/software/libdvi/dvi_timing.c
index cff8edc..82ef50b 100644
--- a/software/libdvi/dvi_timing.c
+++ b/software/libdvi/dvi_timing.c
@@ -14,7 +14,7 @@
 // VGA -- we do this mode properly, with a pretty comfortable clk_sys (252 MHz)
 const struct dvi_timing __dvi_const(dvi_timing_640x480p_60hz) = {
        .h_sync_polarity   = false,
-       .h_front_porch     = 16,
+       .h_front_porch     = 16 + 160,
        .h_sync_width      = 96,
        .h_back_porch      = 48,
        .h_active_pixels   = 640,
```

closed 2024-03-21 by Wren6991