Wren6991/PicoDVI #32: where to ask for help?

bwanaaa, 2022-06-14:
As a beginner learning the details of video, this project is great for teaching the basics of how video is sent to a monitor and working with the limitations of hardware. However, I am still new to software engineering. For example, I dont know how to convert a jpg to a .h file. I dont know how to compile just a single app. I managed to walk through your blog entry and got ALL the examples you made to work. But I am little more than a monkey hacking at a keyboard. How could I compile just ONE app?
ThaNK YOU.

Xiran64, 2022-06-15:
First, you need to decide, if you go for the way of micro phython or c, focus in the documentation about pico. But if you are a begginer, check the first, or arduino

bwanaaa, 2022-06-16:
the examples are written in C. Since I already code for the arduino, it's not that different. I just need to know how to change the project files so as to compile only one example. By making changes and iterating, I can figure out how this thing works. It's the closest thing to a REPL.

Xiran64, 2022-06-17:
I don´t use arduino, so i cannot say to you something about that.

You can put in your favorite searcher:

"pi pico getting started"

The first steps are complicated, but follow the tutorial, and so on.

P.S.:
Pico dvi is completely complicated, you need a lot of things to understand it (I don't do it yet)

Xiran64, 2022-06-17:
The  short answer is very easy:
Press the compile button.
The answer that you need:
All the people that are in here (including me), don't receive one cent for answer easy cuestions or complex cuestions. The creator have good intentions for share to us, a little of their very good ideas, so, if your question is very simple to him, he don't will waste him time, to answer very easy things that you can search for yourself, in very much places, THAT SHOW TO YOU, HOW TO DO IT.

closed 2024-03-21 by Wren6991