Wren6991/PicoDVI #36: Not working with older DVI monitors

IanSB, 2022-06-24:
I've been experimenting with this project on some older DVI monitors and can't get them to display an image. It does work on more modern monitors that have HDMI and DVI connectors. I've been testing with an HP LP2065 and a Dell 2007FPb which are both 1600x1200 4:3 20" monitors with DVI input.

These older monitors also have a problem displaying the output of a raspberry Pi unless the Pi is configured for true DVI output using hdmi_drive=1 in config.txt.

Interestingly, if I put a HDMI splitter between the source and the monitor I can get an image and this works for both PicoDVI and the raspberry Pi when configured for HDMI mode (i.e. NOT using hdmi_drive=1 in config.txt) so it looks like something in the bitstream formatting is upsetting these older DVI monitors.

The splitter I was using was one of these:
https://www.ebay.co.uk/itm/325095286549
I also tried one of these but this one didn't work:
https://www.ebay.co.uk/itm/255345890217
(Maybe the first one is reformatting the bitstream and the other is just buffering it?)

I also tried changing the GPIO drive strength and slew rate settings but that didn't help

Any idea what the problem is?

Wren6991, 2022-06-25:
Sorry, I'm a little mystified by this. The video output by this library is "true DVI", i.e. no HDMI signalling features. Let me know if anything more turns up in your troubleshooting, e.g. you find some video modes that work and some that don't.

Wren6991, 2022-06-25:
Also, you didn't mention what board you are using. Does it power the 5V input to the DVI monitor? Strictly this is required for the monitor to notice the video input, although many monitors are forgiving. It could be that your active splitter is driving the 5V, and your RP2040 board is not.

IanSB, 2022-06-25:
@Wren6991 
>Also, you didn't mention what board you are using

It's the Pimoroni pico dv which has +5v according to the schematic. That appears to be working as the second splitter above is unpowered and it does work with HDMI monitors.

>Let me know if anything more turns up in your troubleshooting, e.g. you find some video modes that work and some that don't.

I've run all the demos and tried changing some of them (the sprite & tile ones) to use other modes and again they all worked on the HDMI monitors but not on the DVI ones.

The most useful thing I've found out so far is that some HDMI splitters will reformat the incoming signal to match the monitor capabilities and that even works with things like the HDMI output of a Sky receiver which doesn't work with a direct connection to the DVI monitors but does work via the first splitter linked above (although obviously no audio).

It seems to me that the possible causes are:
1. Physical - the voltages/timing are a little out of tolerance and that is too much for these older monitors.
As mentioned I tried changing the gpio drive levels and slew rate but that didn't help. Can you confirm that the changes I made would have had the desired effect, I changed the following line in dvi_configure_pad() :
```
	hw_write_masked(
		&padsbank0_hw->io[gpio],
		(<drive> << PADS_BANK0_GPIO0_DRIVE_LSB) | (<slew> << PADS_BANK0_GPIO0_SLEWFAST_LSB),
		PADS_BANK0_GPIO0_DRIVE_BITS | PADS_BANK0_GPIO0_SLEWFAST_BITS | PADS_BANK0_GPIO0_IE_BITS
	);
```
and varied `<drive>` from 0-3 and `<slew>` from 0-1

2. Logical - some part of the bitstream is not entirely correct for the original DVI standard but happens to work with HDMI.

4. The monitor is waiting for the DDC to be read before considering the input as valid which doesn't happen with the current demo software but that seems unlikely. Maybe I will cut the DDC lines on something that works and see if that breaks things.

Can you think of any other possible causes?

IanSB, 2022-06-28:
@Wren6991 

I found the issue:
Some of the pins on the 2 x 20 way surface mount pico socket on the Pimoroni Pico DV board didn't look like they were soldered properly to the PCB. After resoldering all the pins of the socket strips it started working so I'm guessing that one or more of the TMDS lines had a bad connection and that modern HDMI chipsets are a lot more tolerant of that than the older DVI chipsets.

closed 2022-06-29 by IanSB