Wren6991/PicoDVI #44: how to change a picture to *.g  like rgb332.h or rgb535.h

DRAgon734465502, 2023-03-11:
hello

how to change a picture to *.h  like rgb332.h or rgb535.h 

I want to disply my picture, but I can not change them to *.h  hex file

sfranzyshen, 2023-03-11:
I can not confirm this but it looks like either the xxd (found in the vim package) or the hexdump utility was used to turn the binary file into a header file ...

Njolf, 2023-03-19:
> I can not confirm this but it looks like either the **xxd** (found in the vim package) or the **hexdump** utility was used to turn the binary file into a header file ... after poking around ... it seams to build from this script ... https://github.com/Wren6991/PicoDVI/blob/master/software/scripts/packtiles

This appears to work, though mostly for sprites, such as ```https://github.com/Wren6991/PicoDVI/blob/master/software/assets/zelda_mini_plus_walk.png``` - and those then get parsed/displayed by ```https://github.com/Wren6991/PicoDVI/tree/master/software/apps/tiles_and_sprites```. This code can be tinkered with to display very small pictures such as the one in the ```assets``` folder, but this likely won't work if op wants actual images, in which case, something like ```https://github.com/Wren6991/PicoDVI/tree/master/software/apps/vista``` is a better example, though my RPi pico doesn't seem to want to run that properly so i can't experiment on that one to figure it out.

Wren6991, 2024-03-21:
These are generated with the `packtiles` script in `software/scripts/packtiles`. Discussed already in: #13, #37. Please search for existing issues before opening a new one next time.

closed 2024-03-21 by Wren6991