Wren6991/PicoDVI #49: Add support for 240x240 and 480x480 screens

Mice96, 2023-06-06:
Use cases are limited to the small circular displays that are better suited to be controlled by raspberry pi's through HDMI such as
https://www.buydisplay.com/1-28-inch-tft-lcd-display-240x240-round-circle-screen-for-smart-watch
https://www.buydisplay.com/480x480-round-ips-tft-lcd-display-2-1-inch-capactive-touch-circle-screen

Wren6991, 2023-06-08:
Those aren't DVI/HDMI displays.

Do you know the HDMI timings used by the HDMI driver board for the second display?

Mice96, 2023-06-08:
Haven't got a clue in the world since I can't find reliable information about it online and it's still anywhere from 5 to 25 days out. In the mean time I'd say don't worry about it or implement a way for the user to specify a custom resolution so every time someone wants to do something with a non standard resolution it doesn't have to be added to the source code.

Wren6991, 2023-06-08:
> implement a way for the user to specify a custom resolution

You can pass in your own `struct dvi_timing`, so it's already possible to support arbitrary modes without editing any of the source in `libdvi`. See examples of those struct definitions here:

https://github.com/Wren6991/PicoDVI/blob/51237271437e9d1eb62c97e40171fbf6ffe01ac6/software/libdvi/dvi_timing.c#LL15C7-L15C13

Mice96, 2023-06-12:
If I'm not mistaken, even when I create my own struct dvi_timing I am unable to use it. When I attempt to define a display with it throws an error as the custom resolution is not defined in within DVIresolution inside of PicoDVI.h

Wren6991, 2024-03-21:
> PicoDVI.h

That's not a file in this repository, it's part of an Arduino integration written by someone else. Feel free to take it up with the maintainer, but there is nothing here for me to fix.

closed 2024-03-21 by Wren6991