Wren6991/PicoDVI #50: Attemptes to write to flash stall/crash dvi output

Mice96, 2023-06-20:
I've already seen and read both #21 and #35.  #35 just describes a work around which does work for EEPROM but not for littleFS and #21 is just about external flash.  On any attempt to write to flash using littleFS both dvi output and the rest of the code crashes, stalls, locks up idk exactly what. 
Heres the code I used to test this

```ino
#include <PicoDVI.h>
#include <LittleFS.h>
DVIGFX8 display(DVI_RES_320x240p60, true, adafruit_feather_dvi_cfg);
void setup() {  // Runs once on startup
  Serial.begin(9600);
  while (!Serial)
    delay(10);  // Wait for Serial Monitor to open
  Serial.println("Serial started");

  LittleFSConfig cfg;
  cfg.setAutoFormat(true);
  LittleFS.setConfig(cfg);
  
  if (!LittleFS.begin()) {  // Blink LED if insufficient RAM
    Serial.println("LittleFS failed to start");
  } else {
    Serial.println("LittleFS started");
  }

  if (!display.begin()) {  // Blink LED if insufficient RAM
    pinMode(LED_BUILTIN, OUTPUT);
    for (;;)
      digitalWrite(LED_BUILTIN, (millis() / 1000) & 1);
  }
  Serial.println("display started");

  display.setColor(0, 0x0000);  // Color 0: Black
  display.setColor(1, 0xFFFF);  // Color 1: White
  Serial.flush();
}

void loop() {
  display.fillScreen(1);
  delay(3000);
  display.fillScreen(0);
  delay(3000);

  File testFile = LittleFS.open("/Test.txt", "w");
  Serial.println("LittleFS.open");

  testFile.close();
  Serial.println("testFile.close()");

  delay(5000);
  Serial.println("delay(5000);");

  display.swap(false, true);
  Serial.println("display.swap");
}
```

jscrane, 2026-11-13:
Any updates on this? I'm seeing it too, opening a file in `r+` with the Philhower core; same sketch works fine without DVI library...

Board is `adafruit_feather_dvi`. Tried both this lib and the Adafruit fork.

jscrane, 2026-11-14:
OK I guess I didn't read the docs properly :-(

This seems to be a known issue; see the section "Avoiding the Crash" [here](https://learn.adafruit.com/picodvi-arduino-library-video-out-for-rp2040-boards?view=all).