Wren6991/PicoDVI #56: How to run 640x480 60hz with overclocking upto 416Mhz?

xrip, 2024-01-14:
I have ported bunch of emulators like NES/Genesis/Sega/Sega Master system to RP2040 and all of them uses VGA output.
Now im hacking them to add HDMI(DVI) output aswell.

But all of them uses Pi Pico with maximum overclocking > 366Mhz (Pico XT runs at 416Mhz).

It's possible to have pico clocked at 416Mhz and have 640x480 60Hz HDMI output? setting sm_config_clckdiv seems not work or i'm doing something wrong.

Wren6991, 2024-03-21:
No, you need to serialise at exactly the TMDS bit clock, which means the system clock must be an integer multiple of the TMDS bit clock.

In practice this means the TMDS bit clock must be equal to the system clock, because twice the minimum TMDS bit clock is 0.5 GHz, faster than RP2040 will run.

closed 2024-03-21 by Wren6991

xrip, 2024-03-21:
I've solved this with custom, not yours, HDMI driver. It's running with clkdiv 1.5 at 378Mhz 
if you intersted you can take a look at
https://github.com/xrip/pico-megadrive/blob/main/drivers/hdmi/hdmi.c