Wren6991/PicoDVI #60: I´ve can´t hear something...

Xiran64, 2024-03-01:
I´ve made my own test board, and re asign output pines to adapt to my board, i´ve compile the 3 folders with audio extra in the name.
 
But i´ve only see in the tv, the original image, and in the right, some slim lines moving.

In the pdf of ikjordan, i´ve see three new pins, so, i´ve thing, i need it to change mode from dvi to hdmi? can i remap those? Because in my configuration, i´ve use that pines, in tmds signals =(

Xiran64, 2024-03-01:
I´ve see an pwm output pin for olimex, but i can´t see it...

Xiran64, 2024-03-06:
> I´ve see an pwm output pin for olimex, but i can´t see it...

Auto responding me, in olimex board, audio L and R, 22 and 23. I put an amplifier in thoose pins, but no sound.

ikjordan, 2024-03-12:
I see my name mentioned. 
Are you using the code in the fork of PicoDVI that I made? That fork adds a very simple pio, so that the audio pins can be driven at the correct speed for pwm sound. Code still has to be written so that the pwm is driven though. My picozx81 repo gives an example of generating sound on the Olimex board, both over pins 23 and 24, and also over the DVI / HDMI connection.

ikjordan, 2024-03-13:
The code at [https://github.com/ikjordan/PicoDVI](https://github.com/ikjordan/PicoDVI) has been updated today to include more detail in the Readme. There are two branches, [master](https://github.com/ikjordan/PicoDVI) and [audio](https://github.com/ikjordan/PicoDVI/tree/audio)

If you want to generate stereo PWM sound on the Olimex RP2040-PICO-PC DVI board (which uses pins GP27 and GP28 for audio **not 22 and 23**) then you can follow the guidance in the Readme on the master branch. This branch contains a small change to the original PicoDVI to allow the audio PWM to be driven independently of the DVI clock on the Olimex board. Note however that the branch does not contain any PWM sound examples.

If you want generate sound over the DVI / HDMI cable, then you can follow the instructions in the [Readme](https://github.com/ikjordan/PicoDVI/blob/audio/Readme.md) on the audio branch. This branch is based on the audio work of mlorenzati and shuichitakano. There are 3 example programs on the audio branch that play sound over the DVI/HDMI cable. These example are:

1. moon_double_audio: A bit doubled version of the moon demo that includes audio (spoken voice)
2. colour_terminal_audio: Adds spoken voice audio to the colour_terminal demo
3. sprite_bounce_audio: Adds an audio sine wave to the sprite_bounce_demo - caution this is quite loud!

Examples using this code for sound over HDMI or PWM can be found within [https://github.com/ikjordan/picozx81](https://github.com/ikjordan/picozx81)

Xiran64, 2024-03-14:
> I see my name mentioned.
> Are you using the code in the fork of PicoDVI that I made? That fork adds a very simple pio, so that the audio pins can be driven at the correct speed for pwm sound. Code [...]

Thx for answer me ikjordan! I´ts all an honor!

I´ve made a custom board, so the Olimexpc configuration pins, are the more aproach to my board.
But my error was: **sync inverted**.
For that reason i´ve only seen dvi signal, at diferent resolution (695x480) without audio, not the original (640x480) with audio added.
I´ve change dpairs to the oposite, and works! Hdmi signal! Dvi+Audio =).
I tried all three examples with audio added successfully.
So the next step, is to use pwm sounds in a pin or pins, the source of the sound can be a simple *.h file or files contained in the pico flash memory, actioned with a pin with push button or buttons, or at certain time or times =).
I can change physically the working pins ( desolder and resolder it), to the best adapted.

Xiran64, 2024-03-15:
> The code at https://github.com/ikjordan/PicoDVI has been updated today to include more detail in the Readme. There are two branches, [master](https://github.com/ikjordan/PicoDVI) and [...]

Hi again ikjordan, do you thing, it will be possible, to use sounds via hdmi + pwm sounds? All internally in the pico flash, just like a Wii does (sounds on tv + sounds on the wiimote, like effects sounds).

I have read the master branch, but I can't find when I can call a function to play a sound (.h) via pwm on some pin (maybe and we could create a new card with its specific pins, so that they don't 't get in the way internally), and generate sound through both media (hdmi+pwm), or can it be the original
 Wren6991 board, and pwm sounds in some pair of pins of Pmod0 port.

maybe I look like a donkey again =(

ikjordan, 2024-03-16:
@Xiran64 There are no examples within the PicoDVI branches that play sound over PWM. The audio branch includes examples for playing sound over the DVI / HDMI interface. 
Currently the code within [https://github.com/ikjordan/picozx81](https://github.com/ikjordan/picozx81) routes the data to either the DVI/ HDMI interface or to the sound jack connected to GPIO driven by PWM, but you could DMA the data to both PWM and DVI / HDMI so that the same sound is generated through both interfaces simultaneously. The source for this sound data could either be flash or RAM. If the sound data originated in flash, then it may need to be copied into RAM, to avoid disrupting the timing within PicoDVI.

Xiran64, 2024-03-17:
> @Xiran64 There are no examples within the PicoDVI branches that play sound over PWM. The audio branch includes examples for playing sound over the DVI / HDMI interface. Currently the code within [...]

Thx for your answer ikjordan!

I can confirm, i put two diferent sounds, via hdmi (in the tv), and via pwm pin (via an amplifier), both have diferents sounds, and are independent.
I´ve fuse your excelent project with pwm audio reproduction, i´ve recalculate the frecuency, to adjust, to the dvi frequency (only for 640x480) =).

Xiran64, 2024-03-21:
> @Xiran64 There are no examples within the PicoDVI branches that play sound over PWM. The audio branch includes examples for playing sound over the DVI / HDMI interface. Currently the code within [...]

Hi again!

One doubt @ikjordan, about the audio injected in hdmi signal, which characteristics is needed? Because i´ve tried to add audio, but i can only hear noises =(. commodore argentina and my audio only noise

Xiran64, 2024-03-21:
> > @Xiran64 There are no examples within the PicoDVI branches that play sound over PWM. The audio branch includes examples for playing sound over the DVI / HDMI interface. Currently the code within https://github.com/ikjordan/picozx81 routes the data to either the DVI/ HDMI interface or to the sound jack connected to GPIO driven by PWM, but you could DMA the data to both PWM and DVI / HDMI so that the same sound is generated through both interfaces simultaneously. The source for this sound data could either be flash or RAM. If the sound data originated in flash, then it may need to be copied into RAM, to avoid disrupting the timing within PicoDVI.
> 
> Hi again!
> 
> One doubt @ikjordan, about the audio injected in hdmi signal, which characteristics is needed? Because i´ve tried to add audio, but i can only hear noises =(. commodore argentina good and my audio only noise.
> 
> For pwm audio i´ve use uint8_t WAV_DATA, and in your example, i´ve see you use: unsigned char, so i´ve convert my file to the same via xxd method, but only hear noise...

Auto responding me:

It needs to be 8b signed.

Xiran64, 2024-03-27:
Hi again(x2)!

@ikjordan, how can i do, the audio in hdmi do it once? I put at certain time (for example, every 4 secs), alternate with a silent file, but, it repeats the remaining time =(.

ikjordan, 2024-03-27:
> Hi again(x2)!
> 
> @ikjordan, how can i do, the audio in hdmi do it once? I put at certain time (for example, every 4 secs), alternate with a silent file, but, it repeats the remaining time =(.

Currently to play "silence" silence sound packets must be passed. If no sound data is supplied the result is undefined. The author of the ZX Spectrum emulator has raised a pull request so that silence is generated if no sound data is supplied. I'll accept that into "my" repo in the next few days

Xiran64, 2024-03-27:
> Currently to play "silence" silence sound packets must be passed. If no sound data is supplied the result is undefined. The author of the ZX Spectrum emulator has raised a pull request so that [...]

Yep, i´ve generate a silent file, to do that task. But when i need to play only one audio file via hdmi, it repeats all the time to arrive to the silent time.
In pwm audio, i can reproduce a file only once.
I´ve have one idea, measurring the time that audio takes in live time, but it need file for file...

Xiran64, 2024-03-28:
> > > Hi again(x2)!
> > > @ikjordan, how can i do, the audio in hdmi do it once? I put at certain time (for example, every 4 secs), alternate with a silent file, but, it repeats the remaining time =(.
> > 
> > 
> > Currently to play "silence" silence sound packets must be passed. If no sound data is supplied the result is undefined. The author of the ZX Spectrum emulator has raised a pull request so that silence is generated if no sound data is supplied. I'll accept that into "my" repo in the next few days
> 
> Yep, i´ve generate a silent file, to do that task. But when i need to play only one audio file via hdmi, it repeats all the time to arrive to the silent time. In pwm audio, i can reproduce a file only once. I´ve have one idea, measurring the time that audio takes in live time, but it needs to be file for file... Edit: Nope, that would be inaccurate, that will need to occur when the pointer goes to the end (maybe with a flag?), and will be determined from the length, so in the next callback, it need to know, that a "silence", would be used.

UPDATE:

Good news!, it nows play only once, but it needs to restart a variable used internally, because we don´t know which valor it have at that time, and the variable need to say: i´ve terminated speaching, because it only initiallices at 0 at the begging. And second, it depends of wav_len, to start "silence" data, after that, to not corrompt the course of signals.

Xiran64, 2025-04-27:
> [@Xiran64](https://github.com/Xiran64) There are no examples within the PicoDVI branches that play sound over PWM. The audio branch includes examples for playing sound over the DVI / HDMI interface. [...]

Hi again ikjordan!

Do you think is posible to adapt your dvi+audio upgrade to use with rp2350?

Thx in advice!!!

ikjordan, 2025-04-27:
I have :-)

Xiran64, 2025-04-30:
Thx ikjordan!

I´ve found it! 

Thx again!!!

Wren6991, 2025-07-15:
Looks like this was resolved.