Wren6991/PicoDVI #72: Issue with compiling example code

worrallkj, 2024-09-11:
Hello all.

using the Adafruit Feather 2040 RP2040 DVI, Arduino IDE 2.3.2, current version of PicoDVI and associated libraries, and I am getting the following error:

\Documents\Arduino\libraries\PicoDVI_-_Adafruit_Fork\src\libdvi\dvi.c: In function 'dvi_dma_irq_handler':
\Documents\Arduino\libraries\PicoDVI_-_Adafruit_Fork\src\libdvi\dvi.c:194:68: error: 'dma_debug_channel_hw_t' has no member named 'tcr'
  194 |                 while (dma_debug_hw->ch[inst->dma_cfg[i].chan_data].tcr != inst->timing->h_active_pixels / DVI_SYMBOLS_PER_WORD)
      |                                                                    ^

exit status 1

Compilation error: exit status 1

Other RP2040 examples work but anything related to PicoDVI does not compile, all examples flagging the same error. Gone hunting about both the code and various forums but no one seems to have something similar. Happy to be at fault with this but I can't find the issue.

SharifRafid, 2024-09-19:
Facing same issue, did you find a solution?

HisashiKato, 2024-09-20:
Open 'dvi.c' in the ' \Arduino\libraries\PicoDVI_-_Adafruit_Fork\src\libdvi\ ' folder with an editor.

In line 194 of 'dvi.c', change 'tcr' to 'dbg_tcr' in 'dma_cfg[i].chan_data].tcr'.

`- while (dma_debug_hw->ch[inst->dma_cfg[i].chan_data].tcr != inst->timing->h_active_pixels / DVI_SYMBOLS_PER_WORD)`
`+ while (dma_debug_hw->ch[inst->dma_cfg[i].chan_data].dbg_tcr != inst->timing->h_active_pixels / DVI_SYMBOLS_PER_WORD)`

Save the file.

(This text was translated into English using a translation site)

worrallkj, 2024-09-22:
Brilliant - that was the change needed. Works well. tested a couple of the examples and is working. Thanks HisashiKato