Wren6991/PicoDVI #84: Can picodvi+audio port to rp2350?

Xiran64, 2025-04-25:
Hi again!

I want to know, if is posible to emigrate the picodvi+audio from ikjordan to rp2350. In rp forum a friend say me that rp2350 has no more ssi.

Thx in advice!

ikjordan, 2025-04-27:
The [picodvi+audio fork](https://github.com/ikjordan/PicoDVI) libraries were updated to build for the RP2350 in 2024, but some of the example apps were not.

The fork has now been updated and all of the example apps build for the RP2350 using the ARM cores. The one exception is the vista-palette example, which is also excluded from the RP2350 build in this repo.

Xiran64, 2025-04-28:
Thx for answer ikjordan!

Colour terminal app?

I'll rechec it!

Thx again!